sdeigm/uni-meter #194: Voltage & Frequency .. etc. as Input Variables

LukeTheMast3r, 2025-07-10:
Hi sdeigm,

want to extract the second request out of #149 as an enhancement, so that it can be developed seperately.

Request: ->>>
Have seen (within debug mode) that unimeter provides the Frequency and Voltage as output. Is it possibe to configure these values as input data so that they also can be delivered ... same as the power and energy?

Benefit is to get more detailed information out of unimeter to the connected devices. So it is more lieke the original shelly :)

Greetings

sdeigm, 2025-07-11:
That is absolutely possible and is more or less donkey work. As you have seen from the debug output, the system is more or less prepared for this. So which input device would be your first choice?

sdeigm, 2025-07-14:
There are now channels:

voltage
frequency

voltage-l1
frequency-l1
voltage-l2
frequency-l2
voltage-l3
frequency-l3

and sensors

voltage-sensor
frequency-sensor

voltage-l1-sensor
frequency-l1-sensor
voltage-l2-sensor
frequency-l2-sensor
voltage-l3-sensor
frequency-l3-sensor

closed 2025-07-14 by sdeigm