sdeigm/uni-meter #196: Growatt NOAH 2000 shows only zeros from virtual Shelly Pro 3EM

Jasonthefirst, 2025-07-13:
I am attempting to connect my Growatt NOAH 2000 to a virtual Shelly Pro 3EM, which is being simulated by the `uni-meter` Home Assistant add-on. My primary goal is to achieve zero feed-in. While my Growatt NOAH 2000 successfully discovers the virtual Shelly device, the Growatt app consistently displays zero for all power and energy values.

Interestingly, querying the virtual Shelly's RPC endpoint (`http://192.168.0.123/rpc/EM.GetStatus?id=0`) returns the correct, non-zero data, which indicates that `uni-meter` is processing the input from Home Assistant correctly and generating the expected output for the Shelly emulation. The issue appears to be in the communication or data interpretation between the virtual Shelly and my Growatt device.

---

### **To Reproduce**
Steps to reproduce the behavior:
1.  Install and configure the `uni-meter` add-on in Home Assistant.
2.  Set the `input` to `uni-meter.input-devices.home-assistant` and the `output` to `uni-meter.output-devices.shelly-pro3em`.
3.  Configure the Home Assistant sensors as the data source.
4.  Connect the Growatt NOAH 2000 to the same local network.
5.  Allow the Growatt NOAH 2000 to discover the virtual Shelly meter.
6.  Open the Growatt app and observe the data reported from the Shelly meter.

---

### **Expected behavior**
The Growatt NOAH 2000 app should display the real-time power and energy values that are being reported by the virtual Shelly's RPC endpoint.

---

### **Screenshots**

**Growatt App showing zero values:**
[image]

---

### **Configuration**

**`uni-meter` Configuration:**
```hocon
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.home-assistant"

  input-devices {
    home-assistant {
      url = "[http://192.168.0.123:2021](http://192.168.0.123:2021)"
      access-token = "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI2MjllODljOTA1YmY0YTlhOGEwMDViNGJmZDkxYmQ1YyIsImlhdCI6MTc1MjQwNTQ3NSwiZXhwIjoyMDY3NzY1NDc1fQ.BbzTzX1ShozFBmt-TstQe-672n5ZKTN8eZgPCDHHaKA"

      power-phase-mode = "tri-phase"
      energy-phase-mode = "tri-phase"

      power-l1-sensor = "sensor.shelly_em3_channel_a_power_corrected"
      power-l2-sensor = "sensor.shelly_em3_channel_b_power_corrected"
      power-l3-sensor = "sensor.shelly_em3_channel_c_power_corrected"
      
      energy-consumption-l1-sensor = "sensor.shelly_em3_channel_a_energy_returned"
      energy-consumption-l2-sensor = "sensor.shelly_em3_channel_b_energy_returned"
      energy-consumption-l3-sensor = "sensor.shelly_em3_channel_c_energy_returned"

      energy-production-l1-sensor = "sensor.shelly_em3_channel_a_energy"
      energy-production-l2-sensor = "sensor.shelly_em3_channel_b_energy"
      energy-production-l3-sensor = "sensor.shelly_em3_channel_c_energy"
    }
  }

  output-devices {
    shelly-pro3em {
      min-sample-period = 1000ms
      udp-port = 1010
      udp-interface = "0.0.0.0"
    }
  }
}
````

**Virtual Shelly RPC Output (`/rpc/EM.GetStatus?id=0`)**

```json
{
  "id": 0,
  "a_current": 0.86,
  "a_voltage": 230,
  "a_act_power": 197.24,
  "a_aprt_power": 197.24,
  "a_pf": 1,
  "a_freq": 50,
  "b_current": 0.77,
  "b_voltage": 230,
  "b_act_power": 177.48,
  "b_aprt_power": 177.48,
  "b_pf": 1,
  "b_freq": 50,
  "c_current": 0,
  "c_voltage": 230,
  "c_act_power": 0,
  "c_aprt_power": 0,
  "c_pf": 1,
  "c_freq": 50,
  "total_current": 1.63,
  "total_act_power": 374.72,
  "total_aprt_power": 374.72
}
```

-----

### **Additional Context**

The `uni-meter` README states that the Growatt NOAH 2000 is a supported storage system. I suspect the problem might be related to a specific API endpoint or data format that the Growatt device expects from the Shelly Pro 3EM, which might not be fully implemented in the current emulation. The fact that the RPC endpoint shows correct data suggests the core logic of `uni-meter` is working as intended.

```
```

sdeigm, 2025-07-14:
The people that have successfully used the uni-meter in conjunction with the Growatt NOAH 2000 are using the Eco-Tracker as output device. So that seems to work.

Anyhow it would be interesting to figure out what is missing to connect the NOAH to an emulated Shelly Pro 3EM. Since I don't own a NOAH, I am dependent on your or others support to figure that out.

First step would be do increase the logging level of the uni-meter to DEBUG as explained in the README. Then the log file should contain missing HTTP request paths and RPC methods. I assume, that the NOAD tries to query the configuration of the Modbus component which is not yet implemented. 

If the NOAD directly tries to used Modus (without querying the config), that would not show up in the logs since the Modbus component is not implemented at all. Perhaps you can trace the network traffic between the NOAH and the virtual Shelly using Wireshark? That trace should clearly indicate, if the NOAD tries to build up a Modbus connection.

thimo-seitz, 2025-07-14:
The NOAH request `/rpc/EM.GetStatus?esp`, which changes the response to a different format from the uni-meter, than the `/rpc/EM.GetStatus?id=0` request.

sdeigm, 2025-07-15:
I have just tested it, /rpc/EM.GetStatus?esp already delivers the same response as /rpc/EM.GetStatus?id=0. From that perspective it should already work.

Perhaps someone can create a uni-meter log and/or wireshark trace while you are actually trying to connect the uni-meter to the NOAH? The log from @Jasonthefirst just contained data from a running uni-meter which is not queried at all.

thimo-seitz, 2025-07-16:
> I have just tested it, /rpc/EM.GetStatus?esp already delivers the same response as /rpc/EM.GetStatus?id=0. From that perspective it should already work.
>
> Perhaps someone can create a uni-meter log [...]

@sdeigm 
I've wrote something wrong. The NOAH requests the URL `/rpc/Shelly.GetStatus?esp` not `/rpc/EM.GetStatus?esp`:

[image]

sdeigm, 2025-07-16:
Can either of you test with Wireshark if the NOAH starts to query the uni-meter on port 502 using Modbus TCP?

sdeigm, 2025-07-20:
Ok, so my guess/feeling was wrong. The only other thing that is reported through that Shelly.GetStatus call is the enabled/disabled cloud connection. May it be the case, that the NOAH still uses the Shelly cloud values? Is that documented somewhere or do you know from other NOAH/Shelly users if that's the case?

sdeigm, 2025-07-22:
I don't own a real Shelly. Everything is implemented based on the Shelly documentation and other traces I found in the internet. Perhaps I really should buy one which I was initially trying to avoid by building the uni-meter ;-)

I just wanted to add support so that the Shelly reports that it is connected to the cloud and saw that I am already doing so ... also a leftover from other tests. Perhaps that is the problem then? Fortunately this setting is configurable. Can you please test with the following setting in your uni-meter.conf:

```hocon
uni-meter {
  output-devices {
    shelly-pro3em {
      cloud-status {
        enabled = false
        connected = false
      }
    }
  }
}
```

justinh998, 2025-07-23:
Unfortunately, I haven't been home for the last few days.

The Cloud Property doesn't seem to affect anything.
The new properties ‘em:0’ and ‘emdata:0’, on the other hand, seem to be exactly what the battery needs.
Thanks for the quick help.

The real-time data (i.e. everything except the total consumption and feed-in from/to the grid) also works.

However, as far as I can see, this only causes some broken statistics at Growatt and does not affect the actual battery control (why should it?).

I am curious to see if the whole thing runs more stably; the battery has repeatedly lost its connection to the EcoTracker (as far as I can see from the network traffic, this is a Growatt problem).

Translated with [DeepL.com](https://www.deepl.com/?utm_campaign=product&utm_source=web_translator&utm_medium=web&utm_content=copy_free_translation) (free version)

sdeigm, 2025-08-06:
From the RPC call you showed above I can only see, that you uni-meter is getting some data from the input device.

As far as I know (I don't have a NOAH), the RPC method called by the NOAH is  /rpc/Shelly.GetStatus?esp. The fix in version 1.1.14 added two new properties "em:0" and "emdata:0" to the result of that call. The assumption was that these properties are evaluated by the NOAH and that they have been the missing part. 

In the test of @justinh998 the NOAH was displaying some values after these properties had been added. But at the same time there was another feature tested which seemed to have no influence but perhaps it is additionally needed? Can you please add the following entries to your uni-meter.conf and test again?

```hocon
uni-meter {
  output-devices {
    shelly-pro3em {
      cloud-status {
        enabled = false
        connected = false
      }
    }
  }
}
```

Regarding the EcoTracker interface I have no clue why the NOAH has to be reconnected every moring. The interface is absolutely simple, it is just a single HTTP GET request: ```http://<ip>/v1/json```

So why should there be a need for a reconnect every morning? Can you check if the mDNS information stays the same every morning you have to reconnect the EcoTracker?

sdeigm, 2025-08-06:
So can you can confirm, that setting these properties solved the problem?

timniklas, 2025-08-06:
Yes these settings are needed to use the growatt noah 2000 with uni-meter.
Can you add this to the readme?

sdeigm, 2025-08-06:
> In the uni meter configuration I have not set any special options for the Shelly 3 pro EM. Cloud property should therefore be activated for me, as this is standard in uni Meter.

I think that is something I mixed up when building the 1.1.14 version :-( I think in the 1.1.14 version the cloud properties are enabled and in other versions not. @justinh998 can you please recheck this in your environement by executing

http://<ip>/rpc/Shelly.GetStatus

and tell me the cloud status?

I just want to make sure that we get the correct settings ...

timniklas, 2025-08-06:
With this config:
`uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.home-assistant"

  input-devices {
    home-assistant {
      url = "http://192.168.4.243:8123"
      access-token = "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI4ZjM5ZGFmNzg0Mzc0MjJlYjU5YzQ2MjE1MjMxNTdlNCIsImlhdCI6MTc1Mzg4NDI2NiwiZXhwIjoyMDY5MjQ0MjY2fQ.lpSxH4cHgOrctEPnXLtKC7Rl26cOIxF4PUVhvfNU1z8"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"
      power-sensor = "sensor.stromzahler_leistung"
      
      energy-consumption-sensor = "sensor.stromzahler_in"
      energy-production-sensor = "sensor.stromzahler_abgeleitete_einspeisung"
    }
  }

  output-devices {
    shelly-pro3em {
      udp-port = 1010
      udp-interface = "0.0.0.0"
      min-sample-period = 1000ms
      cloud-status {
        enabled = false
        connected = false
      }
    }
  }
}`

the output is:
`{"wifi_sta":{"connected":true,"ssid":"kabellos","ip":"192.168.178.99","rssi":-83},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"15:08","unixtime":1754485680,"serial":1,"has_update":false,"mac":"EE0989C1C431","ram_total":50648,"ram_free":38376,"ram_lwm":32968,"fs_size":233681,"fs_free":174194,"uptime":12119,"temperature":28.08,"overtemperature":false,"tmp":{"tC":28.08,"tF":82.54,"is_valid":true},"modbus":{"enabled":false},"total_power":-11.01,"fs_mounted":true,"em:0":{"id":0,"a_current":-0.02,"a_voltage":230,"a_act_power":-3.67,"a_aprt_power":-3.67,"a_pf":1,"a_freq":50,"b_current":-0.02,"b_voltage":230,"b_act_power":-3.67,"b_aprt_power":-3.67,"b_pf":1,"b_freq":50,"c_current":-0.02,"c_voltage":230,"c_act_power":-3.67,"c_aprt_power":-3.67,"c_pf":1,"c_freq":50,"total_current":-0.06,"total_act_power":-11.01,"total_aprt_power":-11.01},"emdata:0":{"id":0,"a_total_act_energy":305.55,"a_total_act_ret_energy":8.48,"b_total_act_energy":305.55,"b_total_act_ret_energy":8.48,"c_total_act_energy":305.55,"c_total_act_ret_energy":8.48,"total_act":916.65,"total_act_ret":25.44}}`

justinh998, 2025-08-06:
[@justinh998](https://github.com/justinh998) can you please recheck this in your environement by executing
> 
> http:///rpc/Shelly.GetStatus
> 
> and tell me the cloud status?
> 
> I just want to make sure that we get the correct settings ...

In my case, both properties under the cloud array  are true (enabled ,connected).
Maybe Growatt has changed this in the newer firmware (which Martin has). However, I had already tested this in the past and on my firmware it does not matter whether the two cloud properties are true /false.

Since the Shelly simulated by UniMeter is not in the cloud anyway, it would certainly make sense to mark both properties as false you have already said.

sdeigm, 2025-08-08:
Ok, thanks for testing this!

So I will close this issue