sdeigm/uni-meter #201: support for Smartmeter Siemens TD-3511/TD-3512 ?

gremlin1982, 2025-07-14:
would it be possible to implement support for the siemens smartmeters TD-3511/TD-3512 through the Amis-ESP8266 reader?

https://www.mitterbaur.at/amis-leser.html

sdeigm, 2025-07-14:
Is this the output from the REST interface?

[image]

If that's the case, the format should be already supported by the GenericHttp input device. Do you think you can configure that youself? Otherwise I can provide a configuration today in the evening.

sdeigm, 2025-07-14:
Hi, that should be the working uni-meter.conf. Just adjust the IP address:

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.generic-http"
  
  
  output-devices {
    shelly-pro3em {
      udp-port = 1010
    }
  }

  input-devices {
    generic-http {
      url = "http://<amis-ip>/rest"
	  
      power-phase-mode = "tri-phase"
      energy-phase-mode = "tri-phase"
	  
      channels = [{
        type = "json"
        channel = "power-l1"
        json-path = "$.saldo"
      },{
        type = "json"
        channel = "energy-consumption-l1"
        json-path = "$['1.8.0']"
      },{
        type = "json"
        channel = "energy-production-l1"
        json-path = "$['2.8.0']"
      }]
    }
  }  
}
```

sdeigm, 2025-07-14:
Can you please report back if its working? Then I will add it to the samples ...

gremlin1982, 2025-07-15:
yes it works perfect, thank you

closed 2025-07-15 by sdeigm