sdeigm/uni-meter #203: Fehler: unknown channel type: val

osnh1958, 2025-07-18:
Hallo,
ich versuche den Uni-meter zum laufen zu bringen, mit Zugriff auf den Iobroker per simpleAPI.
Die URL in dem Conf-File bringt die richtigen Werte, aber das Programm bringt im Log diesen Fehler:

**25-07-18 12:14:28.744 ERROR org.apache.pekko.actor.typed.Behavior$ - Supervisor RestartSupervisor saw failure [6]: unknown channel type: val**

java.lang.IllegalArgumentException: unknown channel type: val

	at com.deigmueller.uni_meter.input.device.generic_http.GenericHttp.initChannelReaders(GenericHttp.java:152)

	at com.deigmueller.uni_meter.input.device.generic_http.GenericHttp.<init>(GenericHttp.java:64)

	at com.deigmueller.uni_meter.input.device.generic_http.GenericHttp.lambda$create$905c49b3$1(GenericHttp.java:46)

	at org.apache.pekko.actor.typed.javadsl.Behaviors$.$anonfun$setup$1(Behaviors.scala:51)

	at org.apache.pekko.actor.typed.internal.BehaviorImpl$DeferredBehavior$$anon$1.apply(BehaviorImpl.scala:130)

	at org.apache.pekko.actor.typed.Behavior$.start(Behavior.scala:202)

	at org.apache.pekko.actor.typed.internal.RestartSupervisor.restartCompleted(Supervision.scala:404)

	at org.apache.pekko.actor.typed.internal.RestartSupervisor.aroundSignal(Supervision.scala:231)

	at org.apache.pekko.actor.typed.internal.InterceptorImpl.receiveSignal(InterceptorImpl.scala:102)

	at org.apache.pekko.actor.typed.Behavior$.interpret(Behavior.scala:310)

	at org.apache.pekko.actor.typed.Behavior$.interpretSignal(Behavior.scala:271)

	at org.apache.pekko.actor.typed.internal.adapter.ActorAdapter.handleSignal(ActorAdapter.scala:148)

	at org.apache.pekko.actor.typed.internal.adapter.ActorAdapter.aroundReceive(ActorAdapter.scala:114)

	at org.apache.pekko.actor.ActorCell.receiveMessage(ActorCell.scala:590)

	at org.apache.pekko.actor.ActorCell.invoke(ActorCell.scala:557)

	at org.apache.pekko.dispatch.Mailbox.processMailbox(Mailbox.scala:272)

	at org.apache.pekko.dispatch.Mailbox.run(Mailbox.scala:233)

	at org.apache.pekko.dispatch.Mailbox.exec(Mailbox.scala:245)

	at java.base/java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:373)

	at java.base/java.util.concurrent.ForkJoinPool$WorkQueue.topLevelExec(ForkJoinPool.java:1182)

	at java.base/java.util.concurrent.ForkJoinPool.scan(ForkJoinPool.java:1655)

	at java.base/java.util.concurrent.ForkJoinPool.runWorker(ForkJoinPool.java:1622)

	at java.base/java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:165)

Anbei die uni-meter.conf

[uni-meter.txt](https://github.com/user-attachments/files/21316317/uni-meter.txt)

sdeigm, 2025-07-18:
Ach so, falls du einen Marstek Storage verwendest, musst du den power-phase-mode auf "tri-phase" setzten und den dritten Kanal von power-total auf power-l1 ändern. Sonst bleibt der Marstek Storage ewig in der Diagnose hängen, weil er nicht verstellen kann, auf welcher Phase er läuft.

RaBe59, 2025-07-18:
Versuche auch gerade, dass zum Laufen zu bringen. Muss das immer L1 sein, oder die Phase, an der der Marstek tatsächlich hängt?

BTW: Danke für die Arbeit an dem super Tool!

sdeigm, 2025-07-18:
Das ist völlig egal welche der 3 Phasen du nimmst. Der Marstek will nur nicht, dass sich alle 3 Phasen im Gleichtakt bewegen wie es das uni-meter im mono-phase Mode macht.

osnh1958, 2025-07-20:
Vielen Dank sdeigm für die schnelle Antwort, ich konnte es aber erst jetzt ausprobieren.
Das ganze läuft auf einen Raspberry P4 mit 4 GB und aktuellem Raspberry PI OS.
Unimeter läuft über Docker Portainer Installation.

Die Unimeter Variante MonoPhase scheint zu funktionieren, evtl müsste ich noch die Scale Werte anpassen. Über den RPC Aufruf wird keine Aktualisierung alle 5 Sekunden angezeigt, nur über händische Aktualisierung. Die Werte scheinen zu stimmen, wenn ich sie mit den per MQTT abgerrufenen Werten vergleiche. Siehe MonoPhase.png

[image]

[uni-meter_mon.txt](https://github.com/user-attachments/files/21334017/uni-meter_mon.txt)

Beim Versuch das ganze nach Deiner Anleitung auf Tri-Phase umzustellen, kommt folgendes Ergebnis raus:

[image]

[uni-meter_tri.txt](https://github.com/user-attachments/files/21334031/uni-meter_tri.txt)

Der Portainer LOG sieht bei beiden Parametrisierungen gleich aus:

[image]

Beim Versuch per MQTT (über den MQTT Broker im IO Broker zuzugreifen, kommt über den RPC Aufruf diese Meldung:
"power data is currently not available (no data received from the input device)"
Hier die Conf-Datei:

[uni-meter mqtt.txt](https://github.com/user-attachments/files/21334090/uni-meter.mqtt.txt)

Im Log steht auch nichts auffälliges:
25-07-20 09:32:30.759 INFO  uni-meter                - ###################################################################

25-07-20 09:32:30.789 INFO  uni-meter                - # Universal electric meter converter 1.1.13 (2025-07-09 05:22:01) #

25-07-20 09:32:30.790 INFO  uni-meter                - ###################################################################

25-07-20 09:32:30.795 INFO  uni-meter                - initializing actor system

25-07-20 09:32:33.816 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started

25-07-20 09:32:35.553 INFO  uni-meter.controller     - creating ShellyPro3EM output device

25-07-20 09:32:35.711 INFO  uni-meter.controller     - creating MQTT input device

25-07-20 09:32:38.065 INFO  uni-meter.mdns.avahi     - successfully registered mdns service shellypro3em-8ada1ffe4903/_http

25-07-20 09:32:38.092 INFO  uni-meter.mdns.avahi     - successfully registered mdns service shellypro3em-8ada1ffe4903/_shelly

25-07-20 09:32:38.114 INFO  uni-meter.input          - subscribing to topic: iot.smartmeter.sensor.1.obis.1-0:1.8.0.255.value

25-07-20 09:32:38.124 INFO  uni-meter.input          - subscribing to topic: iot.smartmeter.sensor.1.obis.1-0:2.8.0.255.value

25-07-20 09:32:38.126 INFO  uni-meter.input          - subscribing to topic: iot.smartmeter.sensor.1.obis.1-0:16.7.0.255.value

25-07-20 09:32:39.438 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010

25-07-20 09:32:40.369 INFO  uni-meter.input          - MQTT stream connected

25-07-20 09:32:42.188 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80

Wo klemmt es bei mir?

sdeigm, 2025-07-20:
Moin, du hast die tri-phase Config, die ich oben angegeben habe auch nicht richtig kopiert. Im letzten Kanal muss es

`channel = "power-l1"`

heißen, wenn du tri-phase verwendest und nicht 

`channel  = "power-total"

wie du es in deiner Config verwendest die du oben verlinkt hast. Korrigiere das bitte einmal, dann sollte es direkt mit dem Zugriff auf die IoBroker Datenpunkt funktionieren. 

Falls du dennoch MQTT verwenden willst: die Topic-Namen sind falsch. Ich habe das auch erst in den letzten Tagen gelernt, dass die MQTT Broker scheinbar `.` statt `/` im Namen akzeptieren. Die Daten die der MQTT Broker dann aber zurückliefert haben dann aber die `/` im Namen und können deshalb vom uni-meter nicht zugeordnet werden. 

Die Topic Namen müssten irgend etwas wie

iot/smartmeter/sensor/1/obis/1-0:16.7.0.255/value

sein oder so. Das müsstest du mit einem MQTT Browser herausfinden. Aber korrigiere lieber den Channel Wert wie ganz oben angegeben auf power-l1, dann sollte es funktionieren.

closed 2025-08-01 by osnh1958