sdeigm/uni-meter #206: Sunny Portal dont accept the virtuel Shelly Pro 3PM

BirkADH, 2025-07-21:
Hi,

this is my use Case.
I read with Uni-meter and the Generic HTTP Template the power of my go e Charger and simulate a Shelly Pro 3PM.
I do this because the SMA Sunny Portal accept Shelly Pro 3PM and can visualize my wallbox Power at the diagram in Sunny Portal.

Everything is working fine, the Unimeter simulate the Shelly and send the correct data. At my Sunny Portal i can see and bind the virtuell Shelly, but i cant bind a consumer to the virtuell  shelly, you have to do this in the sunny Portal to visulize. With physical shelly it works fine.

I read in other questions that the shelly isnt simulate completely.  I dont know what is missing for my use case but do you think that you expand the virtuell Shelly in the future?

Best Regards
Birk

sdeigm, 2025-07-22:
Perhaps that's possible. To figure out what methods are missing, it would be great if you could create a debug log with the following settings in the `logback.xml` file and post the (relevant) content afterward:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<Configuration status="WARN">
    <appender name="STDOUT" class="ch.qos.logback.core.ConsoleAppender">
        <encoder>
            <pattern>%d{yy-MM-dd HH:mm:ss.SSS} %-5level %-24logger - %msg%n</pattern>
        </encoder>
    </appender>

    <root level="INFO">
        <appender-ref ref="STDOUT" />
    </root>

    <logger name="uni-meter.output"  level="DEBUG" additivity="false">
        <appender-ref ref="STDOUT" />
    </logger>

    <logger name="uni-meter.websocket"  level="DEBUG" additivity="false">
        <appender-ref ref="STDOUT" />
    </logger>
</Configuration>
```

sdeigm, 2025-07-22:
Did you do the binding process during that logging? I don't know the exact procedure with the Sunny Portal, but can you once again "unbind" the virtual Shelly and do the binding again while the above logging is enabled? Or did you already do this and there was no additional output than the one from above? If that's the case please try in again with TRACE log level instead of DEBUG

BirkADH, 2025-07-22:
I did the binding during the process. 
So i repeat it with trace logging. I attched 2 files with 2 binding process. They look almost similary, but in the first on is a log (timecode 25-07-22 17:58:13.607) what are not in the second one, maybe it helps. 
The second try i looked at the time during the process.

at timecode 25-07-22 18:14:00.000 i searched at the Sunny Portal the shellys
at timecode 25-07-22 18:14:45.000 i choose at the Sunny Portal the shellys
at timecode 25-07-22 18:16:15.000 i try to bind the shelly at the Sunny Portal to a consumer, but it didnt work.

I think this are the intresting log. 192.168.178.35 is my Sunny Home Manager 2.0..

`25-07-22 18:14:55.878 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetDeviceInfo()

25-07-22 18:14:55.878 TRACE uni-meter.output         - Shelly.rpcGetDeviceInfo()

25-07-22 18:14:55.878 TRACE uni-meter.output         - ShellyPro3EM.rpcGetDeviceInfo(): {"id":"shellypro3em-0242c0a8b2da","mac":"0242C0A8B2DA","slot":1,"model":"SPEM-003CEBEU","gen":2,"fw_id":"20241011-114455/1.4.4-g6d2a586","ver":"1.4.4","app":"Pro3EM","auth_en":false,"auth_domain":null,"profile":"triphase"}

25-07-22 18:14:55.887 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()

25-07-22 18:14:55.887 TRACE uni-meter.output         - using default power factor 1.0 for /192.168.178.35

25-07-22 18:14:55.887 TRACE uni-meter.output         - using default power factor 1.0 for /192.168.178.35`

[first try.txt](https://github.com/user-attachments/files/21373457/first.try.txt)
[second try.txt](https://github.com/user-attachments/files/21373458/second.try.txt)

sdeigm, 2025-07-22:
Ok, thanks for sending the logs. If 192.168.178.35 is your Home Manager, then it is querying the Shelly every 5 seconds:

```
$ grep onShellyGetStatus second.try.txt
25-07-22 18:14:55.887 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:00.909 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:05.940 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:10.991 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:16.031 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:21.062 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:26.086 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:31.130 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:36.166 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:41.192 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:45.740 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:50.761 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:15:55.778 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:16:00.805 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:16:05.838 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:16:10.856 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:16:15.877 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:16:20.898 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:16:25.944 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:16:30.964 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
25-07-22 18:16:36.008 TRACE uni-meter.output         - ShellyPro3EM.onShellyGetStatus()
```

So it seems if something in this response is not like the Home Manager expects it. You can test the values returned by the uni-meter yourself using the HTTP request `http://<uni-meter-ip>/rpc/Shelly.GetStatus`. 

I will try to google around for some results return by that request ...

sdeigm, 2025-07-22:
I found another sample output in this [forum post](https://forum.arduino.cc/t/shelly-pro-3em-esp8266-json/1152558).

There the Shelly.GetStatus request returns the data in two properties "em:0" and "emdata:0". That's different to what I currently return.

I just prepared a new 1.2.0-SNAPSHOT version which is either available as [Github-Package](https://github.com/sdeigm/uni-meter/packages/2336213) or as docker container sdeigm/uni-meter:1.2.0-SNAPSHOT. That version returns the emeter data in the format mentioned in the arduino forum post. Can you give it a try?

BirkADH, 2025-07-22:
Thanks, but i think i need your help to download or create the docker file. I used github the first time for this.

sdeigm, 2025-07-23:
Ok, so I don't know if you are using  a docker container or if you are directly installing an archive. 

If your directly installing the software you can download the *.tgz archive [here](https://github.com/sdeigm/uni-meter/packages/2336213):

[image]

Then just install it as a normal version with the only difference, that the name of the version is 1.2.0-SNAPSHOT instead of a standard 3 digit version.

If you are using docker just use the image sdeigm/uni-meter:1.2.0-SNAPSHOT instead of sdeigm/uni-meter:latest or sdeigm/uni-meter:1.1.13 in either your docker compose file or docker run command.

Does that answer your question?

PS: if you are using Home Assistant, there is no SNAPSHOT version prepared.

BirkADH, 2025-07-23:
thanks, that was to easy :-) i will have a look in the evening. 
btw. iam Using Docker with a Docker compose File.

BirkADH, 2025-07-23:
I test it and it works :-D . It took a few minutes but then i can bind the virtuell shelly in the Sunny Portal . And the first chargeprocesses were successful.
I would give a longtime feedback after a few weeks and my holiday, if that is ok for you.

Thanks a lot to expand your software for me .