sdeigm/uni-meter #207: failed to parse status response: wrong crc

Ultrahoschie, 2025-07-24:
Hallo,

ich denke ich habe alles richtig konfiguriert, bekomme aber diesen Fehler für die Tibber Integration angezeigt .... 

[08:44:24] INFO: using custom configuration file: uni-meter.conf
25-07-24 08:44:42.375 INFO  uni-meter                - ###################################################################
25-07-24 08:44:42.521 INFO  uni-meter                - # Universal electric meter converter 1.1.13 (2025-07-09 07:16:01) #
25-07-24 08:44:42.526 INFO  uni-meter                - ###################################################################
25-07-24 08:44:42.527 INFO  uni-meter                - initializing actor system
25-07-24 08:44:56.902 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-07-24 08:45:06.506 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-07-24 08:45:07.030 INFO  uni-meter.controller     - creating TibberPulse input device
25-07-24 08:45:28.174 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-07-24 08:45:34.155 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-07-24 08:45:38.371 ERROR uni-meter.input          - failed to parse status response: Timeout
25-07-24 08:45:44.850 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-fa4cb0f7ec2a/_http
25-07-24 08:45:45.087 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-fa4cb0f7ec2a/_shelly
25-07-24 08:50:08.858 ERROR uni-meter.input          - failed to parse status response: wrong crc
25-07-24 08:50:10.126 ERROR uni-meter.input          - failed to parse status response: wrong crc
25-07-24 08:50:11.349 ERROR uni-meter.input          - failed to parse status response: wrong crc
25-07-24 08:50:12.559 ERROR uni-meter.input          - failed to parse status response: wrong crc
25-07-24 08:50:12.575 INFO  uni-meter.input          - omitting further parse errors for the next 24 hours

Ultrahoschie, 2025-07-24:
in der Tibber App und in HA Tibber Lokal  funktiniert alles .... das ist ja das komische

Ultrahoschie, 2025-07-24:
ja node 2 habe ich eingestellt .... 

input-devices {
                        tibber-pulse {
                                  url = "http://192.168.178.24"
                                        node-id = 2
                                              user-id = "admin"

Ultrahoschie, 2025-07-24:
hmm habe es versucht auf tibber local umzustellen ... bekomme aber auch einen fehler .....

25-07-24 10:01:54.645 INFO  uni-meter                - ###################################################################
25-07-24 10:01:54.715 INFO  uni-meter                - # Universal electric meter converter 1.1.13 (2025-07-09 07:16:01) #
25-07-24 10:01:54.725 INFO  uni-meter                - ###################################################################
25-07-24 10:01:54.739 INFO  uni-meter                - initializing actor system
25-07-24 10:02:07.114 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-07-24 10:02:15.336 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-07-24 10:02:15.691 INFO  uni-meter.controller     - creating HomeAssistant input device
25-07-24 10:02:32.644 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-07-24 10:02:39.012 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-07-24 10:02:47.510 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-fa4cb0f7ec2a/_http
25-07-24 10:02:47.689 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-fa4cb0f7ec2a/_shelly
25-07-24 10:35:06.709 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:06.736 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:06.763 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:07.796 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:07.813 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:07.829 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:07.848 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:07.862 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:08.898 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:08.921 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:08.950 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:08.982 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:09.006 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:10.043 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:10.072 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:10.091 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:10.114 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:10.132 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:11.189 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:11.227 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:11.267 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:11.308 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"
25-07-24 10:35:11.351 ERROR uni-meter.input          - failed to parse http response: class java.lang.NumberFormatException - For input string: "unavailable"

config sieht ihmo so aus .... 

uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.home-assistant"

  output-devices {
    shelly-pro3em {
      udp-port = 1010
      udp-interface = "0.0.0.0"
      cloud-status {
      enabled = false
      connected = false
      }
    }
  }

  input-devices {
    home-assistant {
      url = "http://192.168.178.103:8123"
      access-token = "XXXXXXXXX"

      power-phase-mode = "tri-phase"
      energy-phase-mode = "mono-phase"

      power-l1-sensor = "sensor.local_tibber_0100240700ff"
      power-l2-sensor = "sensor.local_tibber_0100380700ff"
      power-l3-sensor = "sensor.local_tibber_01004c0700ff"

      energy-consumption-sensor = "sensor.local_tibber_0100010800ff_in_k"
      energy-production-sensor = "sensor.local_tibber_0100020800ff_in_k"
    }
  }
}

sdeigm, 2025-07-24:
Hallo, so nun komme ich endlich zum Antworten.

> hmm habe es versucht auf tibber local umzustellen ... bekomme aber auch einen fehler .....
>
> 25-07-24 10:01:54.645 INFO uni-meter - ################################################################### [...]

In dem Logausschnitt oben arbeitet das uni-meter von 10:02 bis 10:35 korrekt und du bekommst keine Fehlermeldungen. Dann um 10:35:05 wird dein HomeAssistant Sensor nicht mehr als numerischer Werte sondern als "unavailable" gemeldet. Das macht Home Assistant, wenn es selber keine Daten mehr empfängt. Ich denke deine Verbindung Tibber -> HomeAssistant ist auch nicht 100% stabil.

> was bei der shelly konfig auch komisch ist... er finde den 3em pro sofort in der App als auch in HA ... jedoch ohne die entsprechenden Entitäten, sprich es werden keine daten angezeigt ....

Wie im README beschrieben ist das uni-meter keine vollständige Shelly-Emulation. Ich habe nur nach und nach die von den verschiedenen Storages verwendeten Funktionen implementiert. Was HomeAssistant da genau aufruft weiß ich nicht. Es kann gut sein, dass das mit der kommenden 1.1.14 Version behoben ist, da ich dort die Antwort auf den RPC Aufruf Shelly.GetStatus korrigiert habe.

Aus deiner Tasmota Konfiguration oben entnehme ich (vielleicht liege ich falsch), dass du einen Logarex Stromzähler hast? Falls das so ist, gibt es mit der IR Schnittstelle dieses Zählers wohl Probleme, dass die Dioden altern, siehe z.B. [hier](https://www.photovoltaikforum.com/thread/171145-logarex-lk13be803039-mit-ir-lesekopf-auslesen-anf%C3%A4nger/). Vielleicht kommt es dadurch zu den CRC Fehlern? 

Bei meinem Holley DTZ541 Zähler lieferte der Volkszähler mit dem ich ihn auslese Anfangs auch immer einen CRC Fehler. Das ist aber wirklich ein Bug im Stromzähler. Im Volkszähler wurde dafür extra ein Workaround eingebaut. Es könnte also auch soetwas sein.

 In die kommende 1.1.14 Version habe ich jetzt zusätztlich noch ein Debug Logging für die rohen Tibber Daten eingebaut. Wenn du die Version dann einsetzt kannst du mir ja vielleicht einmal die Daten eines fehlerhaften Tibber Paketes zusenden. Vielleicht fällt mir/uns da dann ja etwas auf.