sdeigm/uni-meter #213: Wie installiere ich das Repository auf HA und wie/wo wird die Configuration-File auf HA angelegt?

LeDocteur82, 2025-07-30:
Aus der aktuellen Beschreibung geht nicht hervor, wie uni-meter simple und smart auf HA integriert wird.
Ich bekomme lediglich das Repository unter Add-on eingebunden und "installiert". Danach ist Game over?!
Habe Samba installiert... habe pyscript installiert... am Ende scheitert alles an der nicht zu findenden Configuration-File. Was muss ich wo und wie noch zusätzlich erledigen?

Hier das permanente Protokoll:

 Add-on: uni-meter
 A universal electrical meter converter
-----------------------------------------------------------
 Add-on version: 1.1.13
 You are running the latest version of this add-on.
 System: Home Assistant OS 16.0  (amd64 / qemux86-64)
 Home Assistant Core: 2025.7.4
 Home Assistant Supervisor: 2025.07.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
custom configuration file uni-meter.conf does not exist
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

LeDocteur82, 2025-07-30:
> You need to create the file yourself. If you have Samba enabled connect with any computer to the homeassistant samba share and in then connect the addon_config share and place a config file in the uni-meter folder like it is mentioned in the installation instructions here https://github.com/sdeigm/uni-meter/blob/main/doc/install/HomeAssistant.md
> 
> <img alt="Image" width="817" height="657" src="https://private-user-images.githubusercontent.com/66540135/472663510-7ee42469-ba6c-4fd4-9b15-41f92ed01e92.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM5MDkyMTgsIm5iZiI6MTc1MzkwODkxOCwicGF0aCI6Ii82NjU0MDEzNS80NzI2NjM1MTAtN2VlNDI0NjktYmE2Yy00ZmQ0LTliMTUtNDFmOTJlZDAxZTkyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzMwVDIwNTUxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUwODQ3ZTYzYTE2YTBkNzZhNTdhY2Q4YTgwNTdhOTlmNGI2ODIyYzJlZjVkNTJiMTFiOGY1ZTViZjJhYmIzM2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0FqHnI_8TZPm1o1SC-ybaRRD5ZhZOilWYbdHv_7giL4">

Thanks! It (maybe) helps a bit. The status in the log remains unchanged.
I'd like a detailed, step-by-step description of the repository for HA!
I'm about to give up... It doesn't seem smart and easy. Perhaps we should introduce know-how levels for the repositories.

sdeigm, 2025-07-31:
The documentation alread said that you have to place the file in that directory:

"The uni-meter is configured using a uni-meter.conf file which should be placed in the /addon_configs/663b81ce_uni_meter directory of your Home Assistant instance. To access that directory, it might be necessary to install additional add-ons which provide access to your Home Assistant instance via Samba or Sftp."

To make it more clear I changed the documentation to:

"The `uni-meter` is configured using a `uni-meter.conf` file which should be placed in the ``/addon_configs/663b81ce_uni_meter`` 
directory of your Home Assistant instance. The installation process is not able to create that file automatically. You 
have to create it **manually**. To do so, it might be necessary to install additional add-ons which provide access to your 
Home Assistant instance via Samba or Sftp."

sdeigm, 2025-08-02:
Have you read and executed this [step](https://github.com/sdeigm/uni-meter/blob/main/doc/install/HomeAssistant.md#mdns-support)?