sdeigm/uni-meter #215: Help with uni-meter.conf for hichi IR v2 MQTT

das399igste, 2025-08-02:
I have a hichi IR v2 for an efr GSM-C8 meter.

uni-meter runs under Docker compose with the following port definition:
```
    ports:
      - 8181:80/tcp
      - 1010:1010/udp
      - 5353:5353/udp
    expose:
      - 80/tcp
      - 1010/udp
      - 5353/udp
```

The connection to the MQTT broker is working. In the uni-meter log file, I find the following entry:

```
received data for MQTT topic tasmota_B95690/SENSOR: 
{
    "Time": "2025-08-02T20:14:52",
    "ENERGY":
    {
        "Power": 181,
        "Total": 70.3012,
        "Supply": 45.2925,
        "wirkleistung_l1": 24,
        "wirkleistung_l2": 76,
        "wirkleistung_l3": 81,
        "Voltage": 235.1,
        "Voltage_L2": 233.6,
        "Voltage_L3": 235.0,
        "Current": 0.25,
        "Current_L2": 0.34,
        "Current_L3": 0.56,
        "Freq": 49.9
    }
}
```
How do I need to set up uni-meter.conf to get reasonable values?

All my attempts so far have only produced junk data.

This is what my uni-meter.conf currently looks like:
```
uni-meter {
  output = "uni-meter.output-devices.eco-tracker"
  
    input = "uni-meter.input-devices.mqtt"
  
  input-devices {
    mqtt {
      url = "tcp://192.168.178.24:1883"
      #username = "username"
      #password = "password"

      power-phase-mode = "tri-phase"
      energy-phase-mode = "tri-phase"

      channels = [{
        type = "json"
        topic = "tasmota_B95690/SENSOR"
        channel = "power-l1"
        json-path = "$..wirkleistung_l1"
        scale = 1.0 # default, can be omitted
      },{
        type = "json"
        topic = "tasmota_B95690/SENSOR"
        channel = "power-l2"
        json-path = "$..wirkleistung_l2"
        scale = 1.0 # default, can be omitted
      },{
        type = "json"
        topic = "tasmota_B95690/SENSOR"
        channel = "power-l3"
        json-path = "$..wirkleistung_l3"
        scale = 1.0 # default, can be omitted
      },{
        type = "json"
        topic = "tasmota_B95690/SENSOR"
        channel = "energy-consumption-total"
        json-path = "$..Total"
      },{
        type = "json"
        topic = "tasmota_B95690/SENSOR"
        channel = "energy-production-total"
        json-path = "$..Supply"
      }]
    }
  }
}
```
I've also tried it with mono-phase, but without success.

Is that even possible? 

It would be great if someone could give me a tip for a correct uni-meter.conf.

Many thanks

das399igste, 2025-08-03:
Vielen Dank für den Hinweis auf die Parameter in der URL zum Abfragen des uni-meter. Ich hatte die URL ohne Parameter aufgerufen und dann nur Platzhalter-Daten gesehen. Mit dem zum output-devices passenden Parametern sieht es nun gut aus.
Auch habe ich in der docker-compose.yml den network_mode: host eingestellt. Die Ports habe ich auf die Standardwerte gesetzt (einen konkurrierenden Docker-Container habe ich dafür umkonfiguriert).

Ich habe gesehen, dass seit Version 1.1.14 (Issues #194) auch voltage und frequency angegeben werden können. Nun habe ich für voltage je Leiter einen Wert (L1, L2, L3), für frequency aber nur einen. Wie würde man das Eintragen bei frequency? Alle drei separat mit demselben Wert oder nur für L1?

Für power und energy kann ich einen phase-mode angeben. 
Kann ich die auch in einer uni-meter.conf mischen?
Beispiel:
```
power-phase-mode = "tri-phase"
energy-phase-mode = "mono-phase"
```
Ich habe das in meiner mal so umgesetzt und es gibt zumindest keine Fehlermeldung im Log.
Unter welchen phase-mode würden voltage und frequency fallen? Oder gibt es dafür separate phase-mode? (Die Hilfeseiten sagen zu voltage und frequency noch nichts ;-) )

Meine aktuelle uni-meter.conf sieht aktuell so aus: [uni-meter.conf.txt](https://github.com/user-attachments/files/21565698/uni-meter.conf.txt)

Ein aktuelles Log liefert folgendes:
```
25-08-03 11:57:03.237 DEBUG uni-meter.input          - received data for MQTT topic tasmota_B95690/SENSOR: {"Time":"2025-08-03T12:57:02","ENERGY":{"Power":-96,"Total":71.9926,"Supply":45.4071,"Wirkleistung_L1":-171,"Wirkleistung_L2":29,"Wirkleistung_L3":45,"Voltage_L1":233.7,"Voltage_L2":236.0,"Voltage_L3":241.0,"Current_L1":0.78,"Current_L2":0.16,"Current_L3":0.35,"Freq":50.0}}

25-08-03 11:57:03.239 TRACE uni-meter.output         - OutputDevice.onNotifyPhasesPowerData()

25-08-03 11:57:03.239 DEBUG uni-meter.output         - power phase 0: PowerData[power=-171.78, apparentPower=-171.78, powerFactor=1.0, current=-0.7350449293966624, voltage=233.7, frequency=50.0]

25-08-03 11:57:03.239 DEBUG uni-meter.output         - power phase 1: PowerData[power=28.84, apparentPower=28.84, powerFactor=1.0, current=0.12220338983050848, voltage=236.0, frequency=50.0]

25-08-03 11:57:03.239 DEBUG uni-meter.output         - power phase 2: PowerData[power=44.65, apparentPower=44.65, powerFactor=1.0, current=0.18526970954356847, voltage=241.0, frequency=50.0]

25-08-03 11:57:03.239 TRACE uni-meter.output         - OutputDevice.onNotifyTotalEnergyData()

25-08-03 11:57:03.239 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=24.0, totalProduction=15.14]

25-08-03 11:57:03.240 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=24.0, totalProduction=15.14]

25-08-03 11:57:03.240 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=24.0, totalProduction=15.14]
```
Auffällig finde ich, dass die für current geliefert Werte von denen abweichen, die im Output angegeben werden. Insbesondere der negative Wert bei Phase 1 irritiert mich. Habe ich noch einen Fehler in der Konfiguration?

Vielen Dank für die Unterstützung und das tolle Projekt.

sdeigm, 2025-08-04:
Das mit dem current ist schon richtig. Den current Wert berechne ich aus der Leistung und der Spannung P = U * I. Das Minus zeigt hier die Richtung des Stroms an. Auf dieser Phase speist du ins Netz ein. Wo der Strom positiv ist beziehst du Leistung aus dem Netz.

Mit der Frequenz solltest du es genau so machen wie du es dir oben schon gedacht hast: frequency-l1 bis l3 alle auf den selben Wert setzen.

power-phase-mode und energy-phase-mode sind beliebig mischbar. Voltage und frequency fallen unter den power-phase-mode.