sdeigm/uni-meter #216: Shelly Pro 3EM connects to app, but always shows offline message

elkos-prints, 2025-08-03:
I have a Hitchi IR sensor which is connected via MQTT to HomeAssistant. Now I want to simulate a Shelly Pro 3EM for an upcoming project with the Marstek Venus E (not bought at the moment). So my first try was to simulate the Shelly and to connect to it through the app to see if everything is working. But let's start step by step.

**1. Installation**
I have installed the uni-meter addon in homeassistant and changed the tcp port from 80 to 1080 and the udp port from 1010 to 2010. The addon ist starting without any issues.

**2. Creating and configuring the uni-meter.conf file**
```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"
  
  input = "uni-meter.input-devices.home-assistant"

  http-server {
    port = 2080
  }
  
  output-devices {
    shelly-pro3em {
      udp-port = 3010
      udp-interface = "0.0.0.0" # default, can be omitted
      port = 4711
    }
  }

  input-devices {
    home-assistant {
      url = "http://192.168.xxx.xxx:8123"
      access-token = "top-secret"
    
      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      power-sensor = "sensor.volkszaehler_mt175_p"
      energy-consumption-sensor = "sensor.volkszaehler_mt175_e_in"
      energy-production-sensor = "sensor.volkszaehler_mt175_e_out"
    }
  }
}
```
After restarting the addon I have got the following Log-files, which look nice to me.

[image]

**3. Testing, if everything works as expected**
Opening the webserver url [http://192.168.xxx.xxx:2080](http://192.168.xxx.xxx:2080)
````
{
  "energyPhase0": {
    "totalConsumption": 1737.96,
    "totalProduction": 17090.32
  },
  "energyPhase1": {
    "totalConsumption": 1737.96,
    "totalProduction": 17090.32
  },
  "energyPhase2": {
    "totalConsumption": 1737.96,
    "totalProduction": 17090.32
  },
  "powerPhase0": {
    "apparentPower": -1728.33,
    "current": -7.51,
    "frequency": 50,
    "power": -1728.33,
    "powerFactor": 1,
    "voltage": 230
  },
  "powerPhase1": {
    "apparentPower": -1728.33,
    "current": -7.51,
    "frequency": 50,
    "power": -1728.33,
    "powerFactor": 1,
    "voltage": 230
  },
  "powerPhase2": {
    "apparentPower": -1728.33,
    "current": -7.51,
    "frequency": 50,
    "power": -1728.33,
    "powerFactor": 1,
    "voltage": 230
  },
  "usageConstraint": "NONE"
}
````

Opening the shelly url [http://192.168.xxx.xxx:4711/rpc/EM.GetStatus?id=0](http://192.168.xxx.xxx:4711/rpc/EM.GetStatus?id=0)
````
{
  "id": 0,
  "a_current": -5.11,
  "a_voltage": 230,
  "a_act_power": -1176,
  "a_aprt_power": -1176,
  "a_pf": 1,
  "a_freq": 50,
  "b_current": -5.11,
  "b_voltage": 230,
  "b_act_power": -1176,
  "b_aprt_power": -1176,
  "b_pf": 1,
  "b_freq": 50,
  "c_current": -5.11,
  "c_voltage": 230,
  "c_act_power": -1176,
  "c_aprt_power": -1176,
  "c_pf": 1,
  "c_freq": 50,
  "total_current": -15.33,
  "total_act_power": -3528,
  "total_aprt_power": -3528
}
````

Looks also nice to me.

**4. Connect to the shelly app**
I have downloaded the official shelly app from the google play store, created an account and tried to connect a new device.
It finds the device, but it does not allow me to connect to it.

[image]

So i tried to enter the shelly ip manually and somehow it seems to let me connect to the device. After clicking "Add device" and entering a device name I saw the device in the app, but marked as offline.

[image]

**5. Alternative way via the Shelly addon in homeassistant**
After that disappointment I tried to add the shelly in homeassistant. It somehow worked, but I don't get any energy values from it.

[image]

What am I doing wrong?

elkos-prints, 2025-08-03:
Ah, I see. Thanks a lot for your answer.
Is there anything else I can test before investing 1300€ in the battery to see if everything will work as expected?

lnxbil, 2025-08-24:
I looked into this as well and found that `/rpc/Shelly.GetStatus` is not providing the ip address, but that will most probably not help. I looked into the URLs that get queried by the app and there is no call to this endpoint or any other endpoints besides `/shelly` (`GET`) and `/rpc` (`POST`). The `POST` has some cryptic value that may be an api token of some sort. It's not base64 decodeable and has always the same prefix for me.

Unless we can intercept and decode the SSL traffic from a real device, we cannot go further from here (e.g. #137)