sdeigm/uni-meter #224: Yet another Tibber Pulse Format?

Olstyle, 2025-08-13:
I succeeded yesterday to connect to my pulse, but I get parsing errors all the time.
Via Browser access to http://_myIP_/data.json?node_id=2 (yes my one node is registered as node 2 for whatever reason) I get the attached data

[Readings.zip](https://github.com/user-attachments/files/21747721/Readings.zip)

All I can see on a very quick glance is that 0x1bB 0x1B 0x1B 0x1B seems to be the delimiter. Is this a new format oder do I do something else wrong?

sdeigm, 2025-08-15:
Sorry for the late reply, but I am currently pretty busy with other things :-(

A timeout indicates, that the message received by the uni-meter is incomplete, the underlying SML library is waiting for additional data. This can normaly not happen with the Tibber Bridge, since it only forwards complete packages via HTTP. If I find some time, I will look into the data but most likely the Tibber Bridge already receives wrong data. Is the reader corretly aligned on your smart meter?  And do you get readings with the tibber App?

Olstyle, 2025-08-15:
The Tibber App is working fine. Can you maybe lead me to some documentation about the binary the data format used by Tibber? Then I can have a look myself.

sdeigm, 2025-08-15:
The Tibber Pulse Bridge currently supported just forwards the binary data received via infrared interface from the electric meter.

The electric meter is normaly sending the data in SML (https://www.bsi.bund.de/SharedDocs/Downloads/DE/BSI/Publikationen/TechnischeRichtlinien/TR03109/TR-03109-1_Anlage_Feinspezifikation_Drahtgebundene_LMN-Schnittstelle_Teilb.pdf?__blob=publicationFile&v=1)

I know that some electrical meters (like mine) have a bug so that the CRC checksum is not calculated correctly. The VzLogger I am using to read the meter data via infrared has a special workaround to accept that wrong checksum.

Olstyle, 2025-08-16:
Thank you.
So two of my datasets are starting as expected with `1b 1b 1b 1b 01 01 01 01` and end as expected with `1b 1b 1b 1b 1a xx xx zz`, but one is much shorter with no end:
``` 
1b 1b 1b 1b 01 01 01 01 76 05 0f fb 5a 98 62 00 62 00 72 63 01 01 76 01 01 05 05 53 c8 de 0b 09 01 49 53 4b 00 04 5a e0 d6 01 01 63 52 b4 00 76 05 0f fb 5a 99 62 00 62 00 72 63 07 01 77 01 0b 09 01 49 53 4b 00 04 5a e0 d6 07 01 00 62 0a ff ff 72 62 01 65 0b 93 72 c4 7a 77 07 81 81 c7 82 03 ff 01 01 01 01 04
```
 So parsing errors for that one would be expected. But this also means my bridge does NOT only forward complete packages (and of course that I should try to better arrange the pulse node).

closed 2025-08-16 by Olstyle