sdeigm/uni-meter #226: Tibber Pulse - "failed to parse status response: Could not decode message"

Ativelos, 2025-08-16:
Moin moin, ich hab den Tibber Pulse mit Hilfe des verlinkten Projekts so lokal integriert bekommen, dass ich die Leistung in HA angezeigt bekomme.
Die uni-meter.conf Datei habe ich im Grunde 1 zu 1 von den hier unter Tibber hochgeladenen übernommen.
Unter http://<bridge-ip>/data.json?node_id=1 bekomme ich folgendes angezeigt:

v6à¦Øb�b�rcvJâH
ISK�öŠc/Ä�v6à¦Ùb�b�rcw
ISK�öŠ�b
ÿÿeJä½tw�`2ISKw�`�ÿ
ISK�öŠw��ÿe�!bRÿeøä<w��ÿbR�S�Ùcö�v6à¦Úb�b�rcqc#Î��ºŽ

Das Uni-Meter Protokoll zeigt nach dem Starten aber das:

s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: uni-meter
 A universal electrical meter converter
-----------------------------------------------------------
 Add-on version: 1.1.15
 You are running the latest version of this add-on.
 System: Home Assistant OS 16.1  (amd64 / qemux86-64)
 Home Assistant Core: 2025.8.2
 Home Assistant Supervisor: 2025.08.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[18:02:04] INFO: using custom configuration file: uni-meter.conf
25-08-16 18:02:07.216 INFO  uni-meter                - ###################################################################
25-08-16 18:02:07.227 INFO  uni-meter                - # Universal electric meter converter 1.1.15 (2025-08-07 20:28:51) #
25-08-16 18:02:07.229 INFO  uni-meter                - ###################################################################
25-08-16 18:02:07.230 INFO  uni-meter                - initializing actor system
25-08-16 18:02:08.690 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-08-16 18:02:09.688 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-08-16 18:02:09.750 INFO  uni-meter.controller     - creating TibberPulse input device
25-08-16 18:02:12.059 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-08-16 18:02:13.103 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-08-16 18:02:13.904 ERROR uni-meter.input          - failed to parse status response: Could not decode message
25-08-16 18:02:15.051 ERROR uni-meter.input          - failed to parse status response: Could not decode message
25-08-16 18:02:16.141 ERROR uni-meter.input          - failed to parse status response: Could not decode message
25-08-16 18:02:17.254 ERROR uni-meter.input          - failed to parse status response: Could not decode message
25-08-16 18:02:18.382 ERROR uni-meter.input          - failed to parse status response: Could not decode message
25-08-16 18:02:18.383 INFO  uni-meter.input          - omitting further parse errors for the next 24 hours

Ist das schon ein bekannter Fehler?

SelfmadestromDE, 2025-08-16:
Bei mir läuft das gut. Ich erhalte ab und zu mal:

failed to parse status response: Timeout

Hier meine Config:

uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"
  
  input = "uni-meter.input-devices.tibber-pulse"

  output-devices {
    shelly-pro3em {
      #min-sample-period = 3000ms
	device {
        mac = "B827EB364242"
        hostname = "shellypro3em-b827eb364242"
  }
 }
}

  input-devices {
    tibber-pulse {
      url = "http://123.456.7.8"
      node-id = 1
      user-id = "admin"
      password = "MeinPasswort"
    }
  }
}

Ativelos, 2025-08-16:
Danke für deine schnelle Antwort.
Ich hab jetzt auch mal den device Abschnitt mit mac und hostname reingepackt, aber da tut sich leider immer noch nichts.
Meine uni-meter.conf:

uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"
  
  input = "uni-meter.input-devices.tibber-pulse"

  http-server {
    port = 80
  }

  output-devices {
    shelly-pro3em {
      device {
        mac = "B827EB364242"
        hostname = "shellypro3em-b827eb364242"
      }
      udp-port = 1010
    }
  }

  input-devices {
    tibber-pulse {
      url = "http://XXX.XXX.X.XXX"
      node-id = 1
      user-id = "admin"
      password = "XXXX-XXXX"
    
    # The Marstek storage needs input data on a single phase. This can be controlled by
    # the configuration options below
    power-phase-mode = "mono-phase"
    power-phase = "l1"
    }
  }
}

(Die Xe natürlich entsprechend ersetzt).

Wie sieht das bei dir unter http://bridge-ip/data.json?node_id=1 aus?

Olstyle, 2025-08-19:
Ich denke du hast ähnliche Schwierigkeiten wie ich:
https://github.com/sdeigm/uni-meter/issues/224
Ein paar Dinge die du tun kannst:

1. Lesekopf vom Pulse neu ausrichten (!)
2. Uni-meter mal direkt abfragen über _IP_/rpc/EM.GetStatus?id=0, bei mir gab es trotz der vielen parse Fehler eine nutzbare Ausgabe
3. Die über http://bridge-ip/data.json?node_id=1 empfangenen Daten mal abspeichern (einfach die Seite speichern) um im hex-editor nachsehen ob das Datenformat passt (siehe mein issue: 1b1b1b1b bis 02 XX yy zz sollten erster bzw. letzter Block sein)

Ativelos, 2025-08-19:
Zu 1: Welchen Log meinst du genau bzw. wo kann ich das sehen? (Hab noch nicht so viel Ahnung). In HA unter Geräte bei der lokalen Tibber Integration unter Logbuch steht zum Beispiel nichts.
Zu 2: Jo, genau. Würde ich auch sagen. 
Um mir nochmal sicherzugehen:

```
 input-devices {
     tibber-pulse {
       url = "http://bridge-ip"
       node-id = 1
       user-id = "admin" <- (Standard)
       password = "XXXX-XXXX" <- das was auf der Bridge steht
```

oder?
Zu 3: Also ich habe mir das jetzt ein paar Minuten angeguckt und das sieht immer gleich lang aus.
Benutzt du das uni-meter auch um einen Speicher zu steuern? Falls ja, welchen?

Olstyle, 2025-08-20:
1. Das Log von uni-meter meine ich. Wenn ich das richtig sehe steht in der HA spezifischen Anleitung auch was dazu wie man das bedient.
2. Ja, das stimmt soweit.
3. Ich benutze es mittlerweile erfolgreich um einen Marstek Jupiter C Plus zu betreiben. Aber als Docker Container mit macvlan und damit eigener IP nur für uni-meter. Bei dir Frage ich mich ob Port 80 für den http-Server funktionieren wird, schließlich teilst du dir die IP mit Home assistant, der ja wahrscheinlich seine eigene UI über Port 80 anbietet. Die HA spezifische Anleitung spricht auch explizit von "HA-IP:uni-meter Port".

Ativelos, 2025-08-20:
1) Ich hab mal diese logback.xml mit dem Inhalt von der Readme erstellt. Dann kommt diese Meldung im uni-meter-Protokoll, wenn DEBUG eingestellt ist:
```
25-08-20 12:16:50.357 DEBUG uni-meter.input          - Received raw data: 1b1b1b1b01010101760536ef81df620062007263010176010105124fd5f50b0a0149534b0003f68a1e01016352cb00760536ef81e0620062007263070177010b0a0149534b0003f68a1e070100620affff65124fd86f747707010060320101010101010449534b0177070100600100ff010101010b0a0149534b0003f68a1e0177070100010800ff65001d290401621e52ff6515fb15770177070100100700ff0101621b520053ff53010101631a5200760536ef81e16200620072630201710163f683001b1b1b1b1a00b693
25-08-20 12:16:50.362 ERROR uni-meter.input          - failed to parse status response: Could not decode message
25-08-20 12:16:51.454 DEBUG uni-meter.input          - Received raw data: 1b1b1b1b01010101760536ef81df620062007263010176010105124fd5f50b0a0149534b0003f68a1e01016352cb00760536ef81e0620062007263070177010b0a0149534b0003f68a1e070100620affff65124fd86f747707010060320101010101010449534b0177070100600100ff010101010b0a0149534b0003f68a1e0177070100010800ff65001d290401621e52ff6515fb15770177070100100700ff0101621b520053ff53010101631a5200760536ef81e16200620072630201710163f683001b1b1b1b1a00b693
25-08-20 12:16:51.458 ERROR uni-meter.input          - failed to parse status response: Could not decode message
```
Und mit TRACE:
```
25-08-20 12:19:52.390 ERROR uni-meter.input          - failed to parse status response: Could not decode message
25-08-20 12:19:52.391 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-08-20 12:19:53.407 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-08-20 12:19:53.407 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-08-20 12:19:53.524 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-08-20 12:19:53.526 TRACE uni-meter.input          - Pulse.onStrictEntity()
25-08-20 12:19:53.527 DEBUG uni-meter.input          - Received raw data: 1b1b1b1b01010101760536ef8401620062007263010176010105124fd6ab0b0a0149534b0003f68a1e01016324f800760536ef8402620062007263070177010b0a0149534b0003f68a1e070100620affff65124fd925747707010060320101010101010449534b0177070100600100ff010101010b0a0149534b0003f68a1e0177070100010800ff65001d290401621e52ff6515fb15770177070100100700ff0101621b520053fdf8010101636b4300760536ef840362006200726302017101636315001b1b1b1b1a003847
25-08-20 12:19:53.531 ERROR uni-meter.input          - failed to parse status response: Could not decode message
```

3) Ich denke schon, dass das der richtige Port sein wird, weil bei laufendem uni-meter Add-on komme ich mit `http://<home-assistant-ip>:<uni-meter-port>` auf diese Seite:

[image]

Wenn ich das Add-on stoppe ist keine Seite erreichbar. Oder habe ich da einen grundlegenden Verständnisfehler?

closed 2026-02-09 by sdeigm