sdeigm/uni-meter #230: Support for HUAWEI Smart Power Sensor DTSU666-H

kili87, 2025-08-22:
As the Sungrow Smart Energy Meter (DTSU666) is already supported, I wanted to ask, if you could also add support for the very similar Huawei version of it - the DTSU666-H.

From GPT research:
While both the Sungrow Smart Energy Meter (DTSU666) and the HUAWEI Smart Power Sensor (DTSU666-H) share a similar model number and are made by the same manufacturer (CHINT), they are not interchangeable and are designed to work exclusively with their respective brand's inverters.

Hence I assume, I can't use the  already existing integration you've built so far.

The data of the smart meter can be querried indirectly via Modbus TCP from the Huawei Smartdongle (which is calling the DTSU via RS485 internally). I'm already doing this in an EVCC setup which calls the power information every few seconds via Modbus.

Please let me know if I can support or provide any further information?

Thanks
Kili

sdeigm, 2025-08-24:
I can try to add support for the DTSU666-H. It think your setup should be like the one mentioned [here](https://www.photovoltaikforum.com/thread/198187-huawei-per-modbustcp-sdongle-an-die-register-des-dtsu666-h/). Do you think that is correct?

If the feature is implemented, you have to support me in testing it :-)

But I think I will not find any time until end of next week ...

kili87, 2025-09-16:
I've tried with the new version, but no success. Error stays with:
`failed to read holding registers at address 37100, quantity 39`

So I've spun up some Modbus TCP debugging tool and tried with different values. I've used this source that documents several registers: https://www.debacher.de/wiki/Sun2000_Modbus_Register 

I've chose register 37113 with 2x 32 bit INT registers and no offset in my settings of the tool, to get the current feed in information from the meter.

[image]

The result is 12415 with a factor of 1000 --> 12,4 kWh
I've double checked with my web interface of the home PV setup and indeed it showed me 12,3kWh (small difference in values as the web interface only polls data every 5 minutes).

[image]

So this leads me to the fact that your Modbus TCP implementation seems wrong. Please try with register 37113. And not sure what other values you can set but quantity 39 seems wrong as well as it is basically 2 registers with 32 bit integer we have to read via option 3.

sdeigm, 2025-09-16:
Ok, I am reading the quantity of 39 registers since this block would contain all values the meter provides (also according to the document you linked - address 37100 to 37138). Do you still get timeout errors with the new version or what other failure reason is reported?

From the above Simply Modbus TCP Client snapshot I can see that the inverter responds very slowly. The single request for a 32bit value (2 Registers) already takes 5 seonds:

[image]

So maybe it ist not possible to query all interesting meter values in a sufficiently small time interval.

I will provide a version which just reads exactly the 37113 register as you did with the modbus client and I will use default values for all the other stuff. So we will see if it works at least basically.

kili87, 2025-09-17:
On my first check it looks good:

uni-meter  | 25-09-17 08:50:41.144 INFO  uni-meter                - ###########################################################################
uni-meter  | 25-09-17 08:50:41.146 INFO  uni-meter                - # Universal electric meter converter 1.1.18-HUAWEI7 (2025-09-17 08:38:29) #
uni-meter  | 25-09-17 08:50:41.146 INFO  uni-meter                - ###########################################################################
uni-meter  | 25-09-17 08:50:41.146 INFO  uni-meter                - initializing actor system
uni-meter  | 25-09-17 08:50:41.503 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
uni-meter  | 25-09-17 08:50:41.710 INFO  uni-meter.controller     - creating ShellyPro3EM output device
uni-meter  | 25-09-17 08:50:41.724 INFO  uni-meter.controller     - creating HuaweiInverter input device
uni-meter  | 25-09-17 08:50:42.004 INFO  uni-meter.mdns.avahi     - successfully registered mdns service shellypro3em-1a6834b35298/_http
uni-meter  | 25-09-17 08:50:42.005 INFO  uni-meter.mdns.avahi     - successfully registered mdns service shellypro3em-1a6834b35298/_shelly
uni-meter  | 25-09-17 08:50:42.005 INFO  uni-meter.input          - connection to the modbus device at 192.168.102.148:502 (unit-id 2) established
uni-meter  | 25-09-17 08:50:42.357 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - voltage A: 229.8
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - voltage B: 233.8
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - voltage C: 232.1
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - current A: 1.45
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - current B: 7.91
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - current C: 9.41
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - active power: 4245.0
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - reactive power: 216.0
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - power factor: -0.97
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - grid frequency: 50.02
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - energy exported: 68216.29
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - energy imported: 16555.66
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - accumulated reactive power: 0.0
uni-meter  | 25-09-17 08:50:47.059 DEBUG uni-meter.input          - meter type: 1
uni-meter  | 25-09-17 08:50:47.060 DEBUG uni-meter.input          - AB voltage: 401.5
uni-meter  | 25-09-17 08:50:47.060 DEBUG uni-meter.input          - BC voltage: 403.5
uni-meter  | 25-09-17 08:50:47.060 DEBUG uni-meter.input          - CA voltage: 400.1
uni-meter  | 25-09-17 08:50:47.060 DEBUG uni-meter.input          - active power A: 236.0
uni-meter  | 25-09-17 08:50:47.060 DEBUG uni-meter.input          - active power B: 1841.0
uni-meter  | 25-09-17 08:50:47.060 DEBUG uni-meter.input          - active power C: 2167.0
uni-meter  | 25-09-17 08:50:47.060 DEBUG uni-meter.input          - meter model detection result: 1
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - voltage A: 229.8
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - voltage B: 233.8
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - voltage C: 232.3
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - current A: 1.76
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - current B: 8.14
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - current C: 9.64
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - active power: 4445.0
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - reactive power: 203.0
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - power factor: -0.975
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - grid frequency: 50.04
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - energy exported: 68216.29
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - energy imported: 16555.66
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - accumulated reactive power: 0.0
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - meter type: 1
uni-meter  | 25-09-17 08:50:48.091 DEBUG uni-meter.input          - AB voltage: 401.5
uni-meter  | 25-09-17 08:50:48.092 DEBUG uni-meter.input          - BC voltage: 403.7
uni-meter  | 25-09-17 08:50:48.092 DEBUG uni-meter.input          - CA voltage: 400.1
uni-meter  | 25-09-17 08:50:48.092 DEBUG uni-meter.input          - active power A: 328.0
uni-meter  | 25-09-17 08:50:48.092 DEBUG uni-meter.input          - active power B: 1896.0
uni-meter  | 25-09-17 08:50:48.092 DEBUG uni-meter.input          - active power C: 2221.0
uni-meter  | 25-09-17 08:50:48.092 DEBUG uni-meter.input          - meter model detection result: 1
uni-meter  | 25-09-17 08:50:49.120 DEBUG uni-meter.input          - voltage A: 229.8
uni-meter  | 25-09-17 08:50:49.120 DEBUG uni-meter.input          - voltage B: 233.8
uni-meter  | 25-09-17 08:50:49.120 DEBUG uni-meter.input          - voltage C: 232.3
uni-meter  | 25-09-17 08:50:49.120 DEBUG uni-meter.input          - current A: 1.76
[... cut ...]
uni-meter  | 25-09-17 08:50:50.152 DEBUG uni-meter.input          - BC voltage: 403.7
uni-meter  | 25-09-17 08:50:50.152 DEBUG uni-meter.input          - CA voltage: 400.1
uni-meter  | 25-09-17 08:50:50.152 DEBUG uni-meter.input          - active power A: 451.0
uni-meter  | 25-09-17 08:50:50.152 DEBUG uni-meter.input          - active power B: 2011.0
uni-meter  | 25-09-17 08:50:50.152 DEBUG uni-meter.input          - active power C: 2336.0
uni-meter  | 25-09-17 08:50:50.152 DEBUG uni-meter.input          - meter model detection result: 1

sdeigm, 2025-09-17:
Yes, all the values are looking reasonable.

- But I guess you are currently exporting power to the grid? So the sign of the power values seems to be wrong.

- Which unit-id are you using and is that the default I should use if no value is provided?

kili87, 2025-09-17:
Based on https://www.debacher.de/wiki/Sun2000_Modbus_Register positive power values mean feed into power grid:
> 0: feed-in to the power grid
< 0: supply from the power grid
So all numbers in your printout are positive as I'm indeed feeding into the grid at the moment.

I'm using Unit-id 2 as I do have two inverters cascaded. Usually the default value is 1 for the first inverter, where the power meter is usually connected as well as the smartdongle.

kili87, 2025-09-17:
If your standard is, that feed in power has a negative value, you need to invert these values for your conversion to Shelly of course.

HolgiB, 2025-09-25:
Just reading along out of interest. I have a Huawei Sun 2000 myself. I read the data from the smart power sensor via Modbus TCP adapter via a Docker container which writes the values to my MQTT-Server. I do not really see a valid purpose to add native support for the DTSU666 directly into uni-meter since you can easily use the MQTT functionality of uni-meter itself to feed it with the power data from the DTSU itself. A simple shell script or Node-RED flow can connect the different MQTT topics as needed.  Just as an idea. Drop me a comment if you need further info how I did it.