sdeigm/uni-meter #236: force decimal places (Marstek B2500 crash)

MtheK, 2025-09-05:
The Marstek B2500 battery recognizes the Shelly Pro 3EM,
but it crashes when one of the power values in the UDP RPC is an integer (it only accepts float values).
Confirmed with the latest B2550 FW V116.6.

Other emulators change the values to enforce decimal places:
https://github.com/tomquist/b2500-meter/blob/c2f6a96794edc97684348bfedad725295a048242/shelly/shelly.py#L27
https://github.com/TheRealMoeder/Energy2Shelly_ESP/blob/24daf43a77f2f600c2cf4ad2d52c04d71196a222/src/main.cpp#L133

Please add an option to force decimal to work around the crappy Marstek Software :)

Thanks!

closed 2025-09-06 by sdeigm

sdeigm, 2025-09-06:
Can you please try the new 1.1.16 version and report back if that fixes the problem. If yes, I will add the B2500 to the storages known to be supported ...

bobybobybob, 2025-09-07:
I use EcoTracker function from your tool and it works very well with the Marstek B2500-D