sdeigm/uni-meter #239: Connection Loss

stuertz, 2025-09-07:
On my Marstek Venus E I get from time to time (very 1 or 2 months) a CT connection loss to the uni-meter.
The Uni-meter reads from my solaredge.

In last both cases this happened around 10pm CEST. (Maybe an update on the solaredge device?)

Workaround was to restart uni-meter.

```
# Universal electric meter converter 1.1.14 (2025-07-24 22:05:22) #
```

I get every 6 seconds in the log:
```log
25-09-07 06:20:38.218 ERROR uni-meter.input          - failed to read holding registers at address 40190, quantity 53
java.util.concurrent.CompletionException: java.util.concurrent.TimeoutException: request timed out after 5000ms
	at java.base/java.util.concurrent.CompletableFuture.encodeThrowable(CompletableFuture.java:332)
	at java.base/java.util.concurrent.CompletableFuture.completeThrowable(CompletableFuture.java:347)
	at java.base/java.util.concurrent.CompletableFuture$UniApply.tryFire(CompletableFuture.java:636)
	at java.base/java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:510)
	at java.base/java.util.concurrent.CompletableFuture.completeExceptionally(CompletableFuture.java:2194)
	at com.digitalpetri.modbus.client.ModbusTcpClient.lambda$sendBufferAsync$3(ModbusTcpClient.java:126)
	at com.digitalpetri.modbus.TimeoutScheduler.lambda$fromScheduledExecutor$0(TimeoutScheduler.java:47)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:572)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:317)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1144)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:642)
	at java.base/java.lang.Thread.run(Thread.java:1583)
Caused by: java.util.concurrent.TimeoutException: request timed out after 5000ms
	at com.digitalpetri.modbus.client.ModbusTcpClient.lambda$sendBufferAsync$3(ModbusTcpClient.java:127)
	... 7 common frames omitted
```
until restart

My config looks like:
```conf
uni-meter {
  http-server {
    port = 8085
  }

  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.solaredge"

  input-devices {
    solaredge {
      address = "192.168.1.64"
      port = 1502
      unit-id = 1
    }
  }

  output-devices {
    shelly-pro3em {
      power-offset-total = 20
      device {
        mac = "B827EB364242"
        hostname = "shellypro3em-b827eb364242"

      }
      port = 8084
      udp-port = 1010
      udp-interface = "0.0.0.0" # default, can be omitted
    }
  }
}
```

Has anyone an idea, or monitored the same behaviour?

sdeigm, 2025-09-08:
Ok, if a restart of the uni-meter solves the problem, it seems as if the entire client connection on uni-meter side becomes invalid. 

I will try to fix that by a complete client reinitialization in case of network failures.

sdeigm, 2025-09-08:
@khaindar, the fix to this issue will just add a complete reinitialization of the modbus client. It does not explain and will not solve your issue, that the uni-meter is not found after a restart. I think we have to investigate this separately.

closed 2025-09-08 by sdeigm

stuertz, 2025-09-09:
> [@sdeigm](https://github.com/sdeigm) [sdeigm-uni-meter-1.csv](https://github.com/user-attachments/files/22219037/sdeigm-uni-meter-1.csv)
> 
> I attached my log, Uni-Meter dis crash again this evening. Maybe the log helps.

This fix is implemented in 1.1.17 - Your log shows 1.1.16 - You may update your installation