sdeigm/uni-meter #241: Addon after Home Assistant start not working

criticallimit, 2025-09-09:
I´m using the addon in Home Assistant. So far it´s working good, but I recognized that after a restart for an update of HA the addon ist not working and there are lot of fault messages in der addon-log. No matter if I set it to "start on boot" or not. As long as the addon is running before HA restarts it rstarts itself with faultmessages.

Anbody else here with the same behavior?

cheers

sdeigm, 2025-09-10:
I just did a Home Assistant update in my environment and the uni-meter is working fine. If I am updating the uni-meter itself I am sometimes getting some "Bind expception, address already in use" error messages. I assume this happens because the ports that have already been used by the previous version of the uni-meter are not yet available again (for whatever reason). But this is automatically solved since the uni-meter tries to bind the ports again and again. After some time it is running without problems, you just have log entries from the unsucessful bind operations.

What error message are there in your log?

criticallimit, 2025-09-10:
this is happening every time I restart Home assistant with uni-meter running:

```
[17:17:54] INFO: using custom configuration file: uni-meter.conf
25-09-08 17:17:58.312 INFO  uni-meter                - ###################################################################
25-09-08 17:17:58.335 INFO  uni-meter                - # Universal electric meter converter 1.1.17 (2025-09-08 04:48:44) #
25-09-08 17:17:58.336 INFO  uni-meter                - ###################################################################
25-09-08 17:17:58.337 INFO  uni-meter                - initializing actor system
25-09-08 17:18:00.874 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-09-08 17:18:01.829 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-09-08 17:18:01.898 INFO  uni-meter.controller     - creating HomeAssistant input device
25-09-08 17:18:03.987 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-09-08 17:18:05.038 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-09-08 17:18:06.326 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-f2cc32efd8c1/_http
25-09-08 17:18:06.375 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-f2cc32efd8c1/_shelly
25-09-08 17:18:41.173 ERROR uni-meter.input          - failed to execute http polling: Tcp command [Connect(192.168.178.10/<unresolved>:8123,None,List(),Some(10 seconds),true)] failed because of java.net.ConnectException: Connection refused
25-09-08 17:19:36.502 ERROR org.apache.pekko.actor.typed.Behavior$ - Supervisor RestartSupervisor saw failure [1]: power data is currently not available (no data received from the input device)
com.deigmueller.uni_meter.common.shelly.RpcException: power data is currently not available (no data received from the input device)
	at com.deigmueller.uni_meter.output.device.shelly.ShellyPro3EM.rpcEmGetStatus(ShellyPro3EM.java:1070)
	at com.deigmueller.uni_meter.output.device.shelly.ShellyPro3EM.createStatus(ShellyPro3EM.java:1331)
	at com.deigmueller.uni_meter.output.device.shelly.ShellyPro3EM.onShellyGetStatus(ShellyPro3EM.java:348)
	at org.apache.pekko.actor.typed.javadsl.BuiltReceive.receive(ReceiveBuilder.scala:225)
	at org.apache.pekko.actor.typed.javadsl.BuiltReceive.receiveMessage(ReceiveBuilder.scala:214)
	at org.apache.pekko.actor.typed.javadsl.AbstractBehavior.receive(AbstractBehavior.scala:74)
	at org.apache.pekko.actor.typed.Behavior$.interpret(Behavior.scala:311)
	at org.apache.pekko.actor.typed.Behavior$.interpretMessage(Behavior.scala:265)
	at org.apache.pekko.actor.typed.internal.InterceptorImpl$$anon$2.apply(InterceptorImpl.scala:67)
	at org.apache.pekko.actor.typed.internal.RestartSupervisor.aroundReceive(Supervision.scala:289)
	at org.apache.pekko.actor.typed.internal.InterceptorImpl.receive(InterceptorImpl.scala:95)
	at org.apache.pekko.actor.typed.Behavior$.interpret(Behavior.scala:311)
	at org.apache.pekko.actor.typed.Behavior$.interpretMessage(Behavior.scala:265)
	at org.apache.pekko.actor.typed.internal.adapter.ActorAdapter.handleMessage(ActorAdapter.scala:141)
	at org.apache.pekko.actor.typed.internal.adapter.ActorAdapter.aroundReceive(ActorAdapter.scala:117)
	at org.apache.pekko.actor.ActorCell.receiveMessage(ActorCell.scala:590)
	at org.apache.pekko.actor.ActorCell.invoke(ActorCell.scala:557)
	at org.apache.pekko.dispatch.Mailbox.processMailbox(Mailbox.scala:272)
	at org.apache.pekko.dispatch.Mailbox.run(Mailbox.scala:233)
	at org.apache.pekko.dispatch.Mailbox.exec(Mailbox.scala:245)
	at java.base/java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:373)
	at java.base/java.util.concurrent.ForkJoinPool$WorkQueue.topLevelExec(ForkJoinPool.java:1182)
	at java.base/java.util.concurrent.ForkJoinPool.scan(ForkJoinPool.java:1655)
	at java.base/java.util.concurrent.ForkJoinPool.runWorker(ForkJoinPool.java:1622)
	at java.base/java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:165)
25-09-08 17:19:36.603 INFO  org.apache.pekko.actor.LocalActorRef - Message [org.apache.pekko.io.Udp$Unbound$] from Actor[pekko://uni-meter/system/IO-UDP-FF/selectors/$a/0#-436383337] to Actor[pekko://uni-meter/system/IO-UDP-FF/selectors/$a/0#-436383337] was not delivered. [1] dead letters encountered. If this is not an expected behavior then Actor[pekko://uni-meter/system/IO-UDP-FF/selectors/$a/0#-436383337] may have terminated unexpectedly. This logging can be turned off or adjusted with configuration settings 'pekko.log-dead-letters' and 'pekko.log-dead-letters-during-shutdown'.
25-09-08 17:19:41.508 ERROR uni-meter.http.port-80   - exception in HTTP server: Ask timed out on [Actor[pekko://uni-meter/user/output#864840628]] after [5000 ms]. Message of type [com.deigmueller.uni_meter.output.device.shelly.ShellyPro3EM$ShellyGetStatus]. A typical reason for `AskTimeoutException` is that the recipient actor didn't send a reply.
25-09-08 17:19:52.052 ERROR uni-meter.output         - UDP server failed: GraphStage [org.apache.pekko.stream.impl.ActorRefBackpressureSinkStage$$anon$1-actorRefWithBackpressureSink] terminated abruptly, caused by for example materializer or actor system termination.
25-09-08 17:19:52.053 INFO  uni-meter.output         - restarting UDP server in PT15S seconds ...
25-09-08 17:19:52.057 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
[... cut ...]
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:500)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:481)

````

sdeigm, 2025-09-10:
Thanks for sharing the log. The failure 

```
19:36.502 ERROR org.apache.pekko.actor.typed.Behavior$ - Supervisor RestartSupervisor saw failure [1]: power data is currently not available (no data received from the input device)
com.deigmueller.uni_meter.common.shelly.RpcException: power data is currently not available (no data received from the input device)
```

should not trigger a restart by the supervisor. That is a bug. I will have a look into this. This happens because your restarting Home Assistant instance does not yet provide the power data to the uni-meter, but the uni-meter is already queried by "something" using the http://<shelly-ip>/status method. Do you have the information which external device uses that HTTP request?

criticallimit, 2025-09-10:
Using uni-meter for my Noah2000. Due to I´m using it "Off-Cloud" I have also GroBro running. Don´t know if that causes this behavior.
HA with GroBro and your uni-meter is running fine once uni-meter is startet after HA is startet.

criticallimit, 2025-09-10:
anaother log after a restart of HA:

s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[14:57:27] INFO: using custom configuration file: uni-meter.conf
25-09-10 14:57:30.979 INFO  uni-meter                - ###################################################################
25-09-10 14:57:30.998 INFO  uni-meter                - # Universal electric meter converter 1.1.17 (2025-09-08 04:48:44) #
25-09-10 14:57:31.000 INFO  uni-meter                - ###################################################################
25-09-10 14:57:31.001 INFO  uni-meter                - initializing actor system
25-09-10 14:57:33.080 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-09-10 14:57:34.149 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-09-10 14:57:34.205 INFO  uni-meter.controller     - creating HomeAssistant input device
25-09-10 14:57:36.262 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-09-10 14:57:37.387 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-09-10 14:57:39.055 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em/_shelly
25-09-10 14:57:39.467 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em/_http
25-09-10 15:05:51.180 ERROR uni-meter.input          - failed to execute http polling: Tcp command [Connect(192.168.178.10/<unresolved>:8123,None,List(),Some(10 seconds),true)] failed because of java.net.ConnectException: Connection refused
25-09-10 15:07:38.753 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:03.100 ERROR uni-meter.input          - failed to execute http polling: Tcp command [Connect(192.168.178.10/<unresolved>:8123,None,List(),Some(10 seconds),true)] failed because of java.net.ConnectException: Connection refused
25-09-10 15:09:29.134 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:30.234 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:31.260 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:32.302 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:33.362 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:34.385 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:35.424 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:36.448 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:37.537 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:38.573 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:39.689 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:40.710 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:41.735 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:42.757 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:43.788 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:44.852 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found
25-09-10 15:09:47.132 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/sensor.stromzaehler_phase1_w failed with status 404 Not Found

criticallimit, 2025-09-10:
yes, thats because I send it. sometimes its like in the first Log where something wrong, another time it is only because of the early start. Don´t know what happens here, just wanna inform you that there could be something wrong, like we see in the first Log.

criticallimit, 2025-09-14:
Here is the log after updating to the new Version:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: uni-meter
 A universal electrical meter converter
-----------------------------------------------------------
 Add-on version: 1.1.18
 You are running the latest version of this add-on.
 System: Home Assistant OS 16.2  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2025.9.3
 Home Assistant Supervisor: 2025.09.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[16:10:19] INFO: using custom configuration file: uni-meter.conf
25-09-14 16:10:22.180 INFO  uni-meter                - ###################################################################
25-09-14 16:10:22.199 INFO  uni-meter                - # Universal electric meter converter 1.1.18 (2025-09-14 15:37:22) #
25-09-14 16:10:22.202 INFO  uni-meter                - ###################################################################
25-09-14 16:10:22.203 INFO  uni-meter                - initializing actor system
25-09-14 16:10:24.467 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-09-14 16:10:25.565 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-09-14 16:10:25.623 INFO  uni-meter.controller     - creating HomeAssistant input device
25-09-14 16:10:27.790 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-09-14 16:10:28.779 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-09-14 16:10:30.185 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-b827eb364242/_shelly
25-09-14 16:10:30.214 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-b827eb364242/_http
25-09-14 16:40:55.073 ERROR uni-meter.input          - failed to execute http polling: Tcp command [Connect(192.168.178.10/<unresolved>:8123,None,List(),Some(10 seconds),true)] failed because of java.net.ConnectException: Connection refused
25-09-14 16:43:54.054 ERROR uni-meter.input          - failed to execute http polling: Tcp command [Connect(192.168.178.10/<unresolved>:8123,None,List(),Some(10 seconds),true)] failed because of java.net.ConnectException: Connection refused
25-09-14 16:44:18.500 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/input_number.shelly_l1 failed with status 404 Not Found
25-09-14 16:44:19.524 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/input_number.shelly_l1 failed with status 404 Not Found
25-09-14 16:44:20.547 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/input_number.shelly_l1 failed with status 404 Not Found
25-09-14 16:44:21.773 ERROR uni-meter.input          - failed to execute http polling: http request to http://192.168.178.10:8123/api/states/input_number.shelly_l1 failed with status 404 Not Found
25-09-14 17:33:40.971 ERROR uni-meter.input          - failed to execute http polling: Tcp command [Connect(192.168.178.10/<unresolved>:8123,None,List(),Some(10 seconds),true)] failed because of java.net.ConnectException: Connection refused
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped